SpinalHDL/SpinalHDL #1710: Scala 3 support

goekce, 2025-04-19:
This issue tracks the Scala 3 support. According to the following discussions, Scala 3 support is currently on hold:

https://github.com/SpinalHDL/SpinalHDL/discussions/1087
https://github.com/chipsalliance/chisel/issues/2908
https://github.com/chipsalliance/chisel/issues/3967

Dolu1990, 2025-04-19:
So overall, the Scala 3 branch of SpinalHDL status is : 
- The spinal.core works,wasn't very difficult to port, just required rewritting the small compiler plugin
- The big pain is in all the little ways scala 3 break backward compatibility, especialy in the spinal.lib
- spinal.lib is half broken, when i tried to port to scala 3, it triggered several scala 3 compiler crashes XD

My opinon so far is that while upgrading to scala 3 could be nice, the down side are potentialy : 
- Split between scala 2 and scala 3 which will hurt maintenance time and split the already small community XD
- Is there a ground breaking benefits to scala 3 ?
- Scala 3 compiler wasn't rock solide at all when i tried => more pain on the user aswell, not sure about today status.
- Scala 3 seems to move away from iDSL on quite a few aspects.

goekce, 2025-04-19:
My first concern was support, but Scala 2.13 will be supported indefinitely: https://www.scala-lang.org/blog/2024/12/16/scala-2-maintenance.html

Some people could be concerned with [new language features of Scala 3](https://docs.scala-lang.org/scala3/new-in-scala3.html), which I cannot evaluate for Spinal as a beginner.

bjourne, 2025-04-28:
> * Split between scala 2 and scala 3 which will hurt maintenance time and split the already small community XD
> 
> * Is there a ground breaking benefits to scala 3 ?

The churn is unfortunate, but people may shy away from learning SpinalHDL since it doesn't support the latest version of the language. Speaking as an outsider, you don't want to invest learning a new HDL that you don't believe have traction.

mrcmry, 2025-04-28:
I can comment as a beginner. I started using Spinal with a very thin theoretical knowledge in verilog (but with a long experience in analog and digital electronic). As someone with a long experience in MCU signal processing I had no experience in Scala (but mostly in C, C++, python and recently Rust).

In this regard Scala 2 vs Scala 3 is not a huge problem, on MCU we are used to toolchains that lag a bit. Also I only use Scala for Spinal so I don't really care. Scala 2+Spinal is much more powerful that system verilog, an upgrade to Scala 3 will not improve this much.

My fear is that tools like scaladoc, vscode, the scala debugger start not to implement improvement for scala 2. Or worst drop support. But since [Scala 3 seems to be able to use Scala 2 libs](https://docs.scala-lang.org/scala3/guides/migration/compatibility-classpath.html) I suppose it's not going to append (not like the python 2 -> 3 transition).

From my beginner (and electrical engineer) perspective, I would prefer a "carefully planed, newer, cleaned" version of SpinalHDL for Scala 3 rather than a quick migration to be up-to-date.

As a beginner sometime when there is to many way to do the same thing for no reason it's a bit overwhelming. I'm sure there are constructs that are here for historical reason and have been replaced by better way to do things.

I think we should start a discussion of what could be simplified or corrected in Spinal for a Scala 3 migration. Progressively we decide what has to change and perhaps mark as deprecated the thing that need to be removed. This way people can adapt and we can migrate slowly the lib. When Scala 3 tools are really rock solid and we are happy with our Spinal "cleanup", we offer an new "Spinal for Scala 3" release.

So people have to do the migration only one time per project and it brings improvement in the language (Scala) and the domain specific language (Spinal).

Readon, 2025-06-17:
For the tool version, I think if a fixed version would work, it is not necessary to boost the version.
Just like verilator, the recent most compatible version is 5.0.22 then we could fixed on it. There might be too much for those tools, but bugs.