SpinalHDL/SpinalHDL #1715: `readAsyncMixedWidth` in commented out in the code but documented in the RTD doc

mrcmry, 2025-04-25:
I'm currently trying to add scaladoc based on the RTD doc.

In the doc here:
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Sequential%20logic/memory.html#mixed-width-ram

`readAsyncMixedWidth` is documented but is commented in the code:

https://github.com/SpinalHDL/SpinalHDL/blob/fa27e2f50b733dde50cd364ef266fc5f2927fd09/core/src/main/scala/spinal/core/Mem.scala#L290

And I did a search, there is no other location (appart from commanded code using it) where this function is present.

Should we not remove it from the RTD doc ?

mrcmry, 2025-04-25:
Note also there is not doc for:

`blackboxAllWhatsYouCan` and `blackboxByteEnables`

If you know the meaning, add it here and I can add it this to my doc PR.

Dolu1990, 2025-04-26:
> Should we not remove it from the RTD doc ?

Yes ^^ 

> blackboxAllWhatsYouCan and blackboxByteEnables
If you know the meaning, add it here and I can add it this to my doc PR.

blackboxAllWhatsYouCan => every memory that SpinalHDL is able to remplace with blackboxes will be replaced with it.
blackboxByteEnables => Every memory which use write port with byte mask will be replaced with an equivalent blackbox. (usefull because because synthesis tool don't support an unified way to infer byte mask in verilog/vhdl)

Thanks !

mrcmry, 2025-04-26:
Fine, thanks. I will remove the commented `readAsyncMixedWidth` from my next scaladoc PR and add the black box description to the doc.

closed 2025-04-26 by mrcmry