SpinalHDL/SpinalHDL #1716: A couple of minor TileLink Cache issues...

jtredux, 2025-04-26:
I'm trying to use the TileLink Cache _without_ the Fiber wrapper and I can't seem to find a set of config parameters that elaborate successfully 

Firstly, I think there's a typo on the downstream Id-width - shouldn't it just be log2Up(generalSlotCount) ? 

Secondly, I seem to get elaboration errors about non-power-of-two state counts for the inflight counter logic when generalSlotCount isn't 8.  If I make (generalSlotCount + ctrlLoopDepth +4) a power of two, I get an error about indexing the slots with an index that is the wrong width. The patch below seems to fix both issues.

diff --git a/lib/src/main/scala/spinal/lib/bus/tilelink/coherent/Cache.scala b/lib/src/main/scala/spinal/lib/bus/tilelink/coherent/Cache.scala
index 70d123ed4..aed1e0ca4 100644
--- a/lib/src/main/scala/spinal/lib/bus/tilelink/coherent/Cache.scala
+++ b/lib/src/main/scala/spinal/lib/bus/tilelink/coherent/Cache.scala
@@ -78,7 +78,7 @@ object Cache extends AreaObject{
     masters = List(M2sAgent(
       name = name,
       mapping = List(M2sSource(
-        id = SizeMapping(0, 2 << log2Up(generalSlotCount)),
+        id = SizeMapping(0, 1 << log2Up(generalSlotCount)),
         emits = M2sTransfers(
           putPartial = SizeRange.upTo(blockSize),
           putFull = SizeRange.upTo(blockSize),
@@ -386,7 +386,7 @@ class Cache(val p : CacheParam) extends Component {
     val fsm = new StateMachine {
       val IDLE, CMD, INFLIGHT, GS = new State()
       setEntry(IDLE)
-      val inflight = CounterUpDown(generalSlotCount + ctrlLoopbackDepth + 4)
+      val inflight = CounterUpDown(1 << log2Up(generalSlotCount + ctrlLoopbackDepth + 4))
       val gsMask = Reg(Bits(generalSlotCount bits))
 
       IDLE.whenIsActive(when(start)(goto(CMD)))

Dolu1990, 2025-04-26:
Hi,

``` 
-        id = SizeMapping(0, 2 << log2Up(generalSlotCount)),
+        id = SizeMapping(0, 1 << log2Up(generalSlotCount)),
```
Hmm if i remember well, the reason of this is 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/tilelink/coherent/Cache.scala#L1245C5-L1245C12
https://github.com/SpinalHDL/SpinalHDL/blob/fa27e2f50b733dde50cd364ef266fc5f2927fd09/lib/src/main/scala/spinal/lib/bus/tilelink/coherent/Cache.scala#L1502

Need the extra bit get that extra information about eviction.

> I get an error about indexing the slots with an index that is the wrong width. 

Where ?

jtredux, 2025-04-26:
Thank - I've reverted it now, the error I get is: 

`error] Exception in thread "main" java.lang.AssertionError: assertion failed: stateCount that is not 2**n is unimplemented when handleOverflow is enabled
[error]         at scala.Predef$.assert(Predef.scala:223)
[error]         at spinal.core.package$.assert(core.scala:545)
[error]         at spinal.lib.CounterUpDown.<init>(Utils.scala:779)
[error]         at spinal.lib.CounterUpDown$.apply(Utils.scala:728)
[error]         at spinal.lib.bus.tilelink.coherent.Cache$$anon$23$$anon$24.<init>(Cache.scala:389)
[error]         at spinal.lib.bus.tilelink.coherent.Cache$$anon$23.<init>(Cache.scala:386)
[error]         at spinal.lib.bus.tilelink.coherent.Cache.$anonfun$flush$1(Cache.scala:380)
[error]         at spinal.core.internals.BooleanPimped.generate(Misc.scala:285)
[error]         at spinal.lib.bus.tilelink.coherent.Cache.<init>(Cache.scala:380)
[error]         at rayshader.RSCSoCTop$$anon$2$$anon$3.<init>(RSCSocTop.scala:333)
[error]         at rayshader.RSCSoCTop$$anon$2.<init>(RSCSocTop.scala:332)
[error]         at rayshader.RSCSoCTop.<init>(RSCSocTop.scala:229)
[error]         at rayshader.RSCSoCTopVhdl$.$anonfun$new$4(RSCSocTop.scala:438)
[error]         at spinal.core.internals.PhaseCreateComponent.$anonfun$impl$319(Phase.scala:2946)
[error]         at spinal.core.fiber.Engine$.$anonfun$create$1(AsyncCtrl.scala:173)
[error]         at spinal.core.fiber.AsyncThread.$anonfun$jvmThread$1(AsyncThread.scala:60)
[error]         at spinal.core.fiber.EngineContext.$anonfun$newJvmThread$1(AsyncCtrl.scala:39)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1`

Which I think is unhappy at line 389 of the cache: 
`   val inflight = CounterUpDown(generalSlotCount + ctrlLoopbackDepth + 4)`

jtredux, 2025-04-27:
Another thing that seems new to me is that the Cache gets rendered to VHDL with assertions in it, which upsets Vivado as they're not synthesisable.

Should  I be using some render-time switch to disable them?

Dolu1990, 2025-05-03:
I pushed val inflight = CounterUpDown(1 << log2Up(generalSlotCount + ctrlLoopbackDepth + 4)) in spinalhdl dev branch :)

About asserts, SpinalConfig(noAssert=true) would prevent them being generated.

Cheers

Dolu1990, 2025-05-08:
The number of id is mostly propertional to the number of slots in the cache.

Else, you wound need axi bridge to remove the ID and fifo them back to the response, it isn't too complicated to implement normaly.
There is a spinalhdl implementation, see Axi4IdRemover

closed 2025-05-25 by jtredux