SpinalHDL/SpinalHDL #1770: Nested PackedWordBundle unpack extraction range problem

oletf, 2025-07-22:
Greetings,

I've just stumbled on this problem, tried to make a minimal reproduction 

```scala
case class TestPb() extends PackedWordBundle(32 bits) {
    val u8  = UInt(8 bits).packFrom(3).inWord(1)
    val u16 = UInt(32 bits).packFrom(0).inWord(2)
  } // BitsWidth = 40 , PackedWidth = 96

  case class TestNestPb() extends PackedWordBundle(32 bits) {
    val pb  = TestPb()
    val u32 = Bits(32 bits).packFrom(0).inWord(3)
  } // BitsWidth = 72 , PackedWidth = 128

  val tpb = TestPb()
  tpb.unpack(B(123456, 96 bits))  // no problem
  val tnpb = TestNestPb()
  tnpb.unpack(B(123456, 128 bits)) // triggers following error
```
last line 
```
Static bits extraction (42 downto 35) is outside the range (39 downto 0) of Bits ## Bits at
    at spinal.lib.PackedBundle.$anonfun$unpack$2(PackedBundle.scala:118)
    at spinal.lib.PackedBundle.$anonfun$unpack$2$adapted(PackedBundle.scala:104)
    at spinal.lib.PackedBundle.unpack(PackedBundle.scala:104)
    at spinal.lib.PackedBundle.unpack(PackedBundle.scala:101)
    at spinal.lib.PackedBundle.$anonfun$unpack$2(PackedBundle.scala:120)
    at spinal.lib.PackedBundle.$anonfun$unpack$2$adapted(PackedBundle.scala:104)
    at spinal.lib.PackedBundle.unpack(PackedBundle.scala:104)
    at spinal.lib.PackedBundle.unpack(PackedBundle.scala:101)
```

I think I understand it seems like it tries to extract from the position of the PackedBits in the range of the non-PackedBits.

I tried figuring the problem out, but not sure if it's actually coming from [`PackedWordBundle.inWord`](https://github.com/SpinalHDL/SpinalHDL/blob/1d490d4f4927c2301bd7ebccdeb266b547edf8b4/lib/src/main/scala/spinal/lib/PackedBundle.scala#L211) method which breaks some mappings for the `unpack`, or the [`unpack`](https://github.com/SpinalHDL/SpinalHDL/blob/1d490d4f4927c2301bd7ebccdeb266b547edf8b4/lib/src/main/scala/spinal/lib/PackedBundle.scala#L103) method itself.

Thanks forward !

Dolu1990, 2025-07-28:
@oletf I personnaly do not remember too much about it.
@kleinai @dokleina were more involved :D

dokleina, 2025-08-10:
Sorry, I've been busy at work and dealing with moving apartments recently.

@kleinai has not been an active contributor for some time and he won't be back ☹️ 

It looks like I neglected to account for the _packed_ size of `PackedBundles` when added them to a `PackedBundle`'s internal mapping. I also neglected to use their _packed_ size when determining how many bits they should have to pack nested `PackedBundles`.

See if this fix works https://github.com/dokleina/SpinalHDL/commit/40a8d09992882c32fe2886d0bd06703955b63dab

oletf, 2025-08-11:
Thanks a lot for the fix !

Can confirm https://github.com/dokleina/SpinalHDL/commit/b8e5ebeb3e69aa7956113f526bbb9f97ff9084e4 works as expected for my case !

There's still a weird edge case I didn't really understand :
using the minimal example I've used above :
the original :
```scala
  case class TestNestPb() extends PackedWordBundle(32 bits) {
    val pb  = TestPb()
    val u32 = Bits(32 bits).packFrom(0).inWord(3)
  } // BitsWidth = 72 , PackedWidth = 128
```
gives:
```vhdl
  zz_tnpb_pb_u8 <= pkg_extract(bits,95,0);
  tnpb_pb_u8 <= unsigned(pkg_extract(zz_tnpb_pb_u8,42,35));
```

but using :
```scala
  case class TestNestPb() extends PackedWordBundle(32 bits) {
    val pb  = TestPb().packFrom(0).inWord(0)
    val u32 = Bits(32 bits).packFrom(0).inWord(3)
  } // BitsWidth = 72 , PackedWidth = 128
```
gives the following with some weird padding, which gets the real value out of the correct range :
```vhdl
  zz_tnpb_pb_u8 <= pkg_cat(pkg_extract(bits,39,0),pkg_stdLogicVector("00000000000000000000000000000000000000000000000000000000"));
  tnpb_pb_u8 <= unsigned(pkg_extract(zz_tnpb_pb_u8,42,35));
```

It's what I originally had, I just removed the extra `.packFrom(0).inWord(0)` on the nested `PackedBundle` to test, and it solved the problem.

dokleina, 2025-09-27:
I haven't forgotten about this! I've been busy with life stuff the past month or so.

I'm looking into the edge case you described and would ideally like to have it fixed along with https://github.com/dokleina/SpinalHDL/commit/b8e5ebeb3e69aa7956113f526bbb9f97ff9084e4.

PR #1821 (2025-10-06): Nested PackedWordBundle unpack fix

Closes #1770

# Context, Motivation & Description

This issue stems from PackedWordBundle (or really PackedBundle) needing to call to `packed`/`getPackedWidth` instead of `asBits`/`getBitWidth` to get the intended bits and their bit width.

The `packed` and `unpack` methods were neglecting to call `getPackedWidth` on all nested PackedBundles. This PR adds awareness of PackedBundles when `pack`ing and `unpack`ing

# Impact on code generation

Fixes some unintended verbosity when using nested PackedBundles. Example:

```scala
  case class TestNestPb() extends  [... cut]
closed 2025-10-06 by Dolu1990