SpinalHDL/SpinalHDL #1778: How do I connect a `FiberPlugin` to the outside world?

Syndace, 2025-08-04:
Hi!

I am trying to build a component that makes use of the fiber plugin system and I can't figure out how to connect any of the plugin logic with the top level bundle. Here is my first attempt:

```Scala
import spinal.core._
import spinal.lib.misc.plugin._

case class TestBundle() extends Bundle {
  val test = out Bool()
}

class TestPluginHost extends Component {
  val host = new PluginHost()
}

class TestPlugin extends FiberPlugin {
  val io = TestBundle()

  val logic = during build new Area {
    io.test := True
  }
}

case class TestTop() extends Component {
  val io = TestBundle()

  val pluginHostComponent = new TestPluginHost()
  val testPlugin = new TestPlugin()
  testPlugin.setHost(pluginHostComponent.host)

  testPlugin.io <> io
}

object TestTopVerilog extends App {
  Config.spinal.generateVerilog(TestTop())
}
```

This gives me `AUTOCONNECT FAILED, mismatched directions`.

My second attempt was replacing `TestPlugin` with this:

```Scala
class TestPlugin extends FiberPlugin {
  val io = TestBundle()

  val logic = during build new Area {
    val logicIo = TestBundle()

    logicIo.test := True
  }

  io <> logic.get.logicIo
}
```

This gives me `! SpinalHDL async engine is stuck !`.

I don't have any other ideas. How do I connect plugin logic to an outside bundle?

Also thank you so much, SpinalHDL is incredibly cool and fun :D

Syndace, 2025-08-04:
Hmm, I think `during` only exists in `FiberPlugin`, but my top is no plugin

Dolu1990, 2025-08-04:
Ahh you are right.

outside plugins you can do Fiber build new Area{ ... }

can be imported via 
import spinal.core.fiber._

Syndace, 2025-08-05:
Thank you!! The following works:

```Scala
class TestPlugin extends FiberPlugin {
  val logic = during build new Area {
    val logicIo = TestBundle()

    logicIo.test := True
  }
}

case class TestTop() extends Component {
  val io = new TestBundle()

  val pluginHostComponent = new TestPluginHost()
  val testPlugin = new TestPlugin()
  testPlugin.setHost(pluginHostComponent.host)

  val fiberLogic = Fiber build new Area {
    io <> testPlugin.logic.get.logicIo
  }
}
```

closed 2025-08-05 by Syndace