SpinalHDL/SpinalHDL #1792: Explain the code

looongbin, 2025-08-31:
Hello,
    Can you explain the purpose of this code? I am not sure if I should copy this code into APLIC's mapper.

https://github.com/SpinalHDL/SpinalHDL/blob/474354e1b991d4b854310f3ebb4c11dd069f6586/lib/src/main/scala/spinal/lib/misc/plic/PlicMapper.scala#L121-L131

Dolu1990, 2025-09-03:
Hi,

The goal of this hardware is to make every access to the PLIC at least 3 cycles long.
I think this is to ensure that changes made by one access are visible to the following access (there is a bit of pipelining in PLIC logic)

If i'm right about this, then this implementation isn't great i would say (but functional), it would have been better to set a counter after each access is done, have that counter always decrement back to zero, and only block accesses until that counter isn't 0

looongbin, 2025-09-05:
I understand. Thanks for your reply!

closed 2025-09-05 by looongbin