SpinalHDL/SpinalHDL #1795: SimPublic() not working on specific signal

MahirAbbas, 2025-09-06:
Weird issue. One week ago I didn't have this issue, but now I do?
The SimPublic tag is applied (I've checked in debugging). Specifically on the ``` readHere.pc``` signal. ```PC_delayed ``` is a Payload signal.
```Scala
case class frontEnd() extends Component {
  val pipeline = new StageCtrlPipeline()
  val pc = new PC(pipeline.ctrl(0))
  // pc.PC_cur.simPublic()
  val fetch = Fetch(pipeline.ctrl(1), addressWidth = 64, dataWidth = 32)
  val ram = new UnifiedRam(addressWidth = 64, dataWidth = 32, idWidth = 16)
  val readStage = pipeline.ctrl(2)
  val readHere = new readStage.Area {
    val instr = up(INSTRUCTION)
    val pc = up(Fetch.PC_delayed)
    instr.simPublic()
    pc.simPublic()
  }

  pc.exception.setIdle()
  pc.jump.setIdle()
  pc.flush.setIdle()
  ram.io.reads.cmd << fetch.io.readCmd.cmd
  fetch.io.readCmd.simPublic
  ram.io.reads.rsp >> fetch.io.readCmd.rsp
  fetch.io.readCmd.simPublic()
  pipeline.build()
}
```

```

UNACCESSIBLE SIGNAL : ??? :  UInt[64 bits]) isn't accessible during the simulation.
- To fix it, call simPublic() on it during the elaboration.
[Error] Simulation failed at time=200
Exception in thread "main" java.lang.Exception
        at spinal.sim.SimError$.apply(Misc.scala:12)
        at spinal.core.sim.package$.spinal$core$sim$package$$btToSignal(package.scala:70)
        at spinal.core.sim.package$.spinal$core$sim$package$$getBigInt(package.scala:148)
        at spinal.core.sim.package$SimBitVectorPimper.toBigInt(package.scala:554)
        at test.frontend.test_fetch$.$anonfun$new$6(test_fetch.scala:89)
        at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:192)
        at test.frontend.test_fetch$.$anonfun$new$3(test_fetch.scala:78)
        at test.frontend.test_fetch$.$anonfun$new$3$adapted(test_fetch.scala:44)
        at spinal.core.sim.SimCompiled.$anonfun$doSimApi$2(SimBootstraps.scala:626)
        at spinal.sim.SimManager.threadBody$1(SimManager.scala:246)
        at spinal.sim.SimManager.$anonfun$run$1(SimManager.scala:249)
        at spinal.sim.SimThread.$anonfun$jvmThread$1(SimThread.scala:90)
        at spinal.sim.JvmThread.run(SimManager.scala:51)
```
snippet of code for simulation.
```

      for(i <- 0 to (instructions.length + 1)) {
        dut.clockDomain.waitSampling(1)
        if(i >= 2) {
          var idx = i-2
          if(idx >=15) {
            idx=15
          }
          val(address, data) = instructions(idx)
          // println(s"Data should be ${data.toLong.toHexString}, got ${dut.readHere.instr.toBigInt.toLong.toHexString} at ${dut.readHere.pc.toBigInt.toLong}")
          // println(s"Data should be ${data.toLong.toHexString}, got ${dut.readHere.instr.toBigInt.toLong.toHexString}")
          println(s"Data should be ${data.toLong.toHexString}, got ${dut.readHere.instr.toBigInt.toLong.toHexString}")
          print(s"at ${dut.readHere.pc.toBigInt.toLong}")
          // print(s"at ${dut.readHere.pc}")
        }
```

MahirAbbas, 2025-09-06:
This code was working a week ago, but now I get this inaccessible signal issue. Scala version 2.13.4, SpinalVersion 1.12.3

Dolu1990, 2025-09-10:
Hi,

I tried to reproduce the issue with : 

```scala
object Brarwww666 extends App{
  case class frontEnd() extends Component {
    val pipeline = new StageCtrlPipeline()
    val PC = Payload(UInt(32 bits))
    pipeline.ctrl(0).up(PC) := in.UInt(32 bits)
    val readStage = pipeline.ctrl(2)
    val readHere = new readStage.Area {
      val pc = up(PC)
      pc.simPublic()
    }

    pipeline.build()
  }

  SimConfig.doSim(new frontEnd()){dut =>
    sleep(10)
    println(dut.readHere.pc.toBigInt)
  }
}
``` 

But so far it work on my side (scala 2.12 / spinalhdl dev)

Can you try that snippet on your end ?